Team254/cheesy-arena #12: Can't generate match schedule (2015 release)

TheBassett, 2015-09-30:
Added 25 teams to the teams list then tried to generate a match schedule (as shown in the screenshot):
[image]

After clicking the "Generate Schedule" button it throws up the "Incomplete or invalid schedule block parameters specified." error message. I followed along with the demo video so I don't believe I missed any steps - correct me if I'm wrong.

bkeeneykid, 2015-11-08:
Also getting this issue. Is this in the processor of being fixed or is it even acknowledged?

patfair, 2015-11-09:
I wasn't able to reproduce this issue, but it sounds like something JavaScript-related. What browser is this being seen on?

TheBassett, 2015-11-09:
I originally tested this on chrome. I tried the same setup I had before (CA saved the teams list) and it works now - it was problably just a bug in chrome. @bkeeneykid If you're using Chrome, is it up to date?

bkeeneykid, 2015-11-09:
I now use Safari 8 on Mac. I'll try Chrome later.

phil-lopreiato, 2015-11-09:
@patfair I can reproduce the issue on Firefox 42/Fedora 23. Didn't see any JS errors, but I'm not familiar enough to debug thoroughly. It does however work properly for me on Chrome 46

patfair, 2015-11-16:
I figured out the root cause -- Firefox can't parse dates that use a hyphen as the separator. Officially only Chrome is supported, but I'll fix this eventually.

rich-coe, 2016-02-19:
I reproduced this issue on linux with chrome Version 48.0.2564.109 (64-bit)
I tried the same chrome version on windows and it works.

commit 11b6b5f8 (2017-09-03): Refactor schedule JS to be less buggy (fixes #12).
closed 2017-09-03 by patfair