Team254/cheesy-arena #14: Cant get the 2016 FRC Driver Station To connect to the FMS

Chase2224, 2016-02-04:
I am currently on the FRC team 4778 and we are having trouble with the 2016 driver station by getting it to connect with Cheesy Arena. It works with the 2015 version but the latest one does not work.

AustinSchuh, 2016-02-04:
Did you set your driver station to 2014 FMS mode?

Chase2224, 2016-02-05:
There is no 2014 FMS mode in 2016 driver station only 2015 and 2014 with no fms

patfair, 2016-02-05:
If that's the case, then unfortunately there's no way to make it work until the 2015/2016 DS protocol is published or a 2014 FMS mode is added.

Chase2224, 2016-02-05:
So what you are saying is that as long as we are using the Cheesy Arena 2014 we can not move on to the 2016 Driver Station.

patfair, 2016-02-05:
That's correct. You may be able to use the 2015 Driver Station to talk to your 2016 robot using 2014 FMS mode, but a solution for using Cheesy Arena with the 2016 Driver Station likely won't exist until Cheesy Arena is updated for the 2016 game (after the official season ends).

phil-lopreiato, 2016-02-05:
Does anyone know if the communication protocol for 2016 the same as 2016? I think it is, but I have no confirmation on that fact

If so, [here](https://github.com/WinT-3794/LibDS/) is a C++ implementation of the 2015 protocol that could be useful when adding 2016 support.

AustinSchuh, 2016-02-05:
@phil-lopreiato That looks like the DS protocol, not the FMS protocol (which is different)

patfair, 2016-02-05:
There was an updated FMS Lite released last fall, so one could probably reverse-engineer the 2015 FMS protocol from that. No idea though whether 2015 and 2016 are identical.

Cyken-Zeraux, 2016-02-17:
The 2015 Offseason DS has the older options. 

https://usfirst.collab.net/sf/frs/do/listReleases/projects.wpilib/frs.driver_station

We had to choose FMS Offseason over chezy this year due to the issues of having to set static IP's and such. Hopefully next year we'l get chezy back, because Offseason is very poorly built for this.

Bennett2977, 2016-02-18:
The 2015 Offseason DS works with the 2016 Labview RTE, and works with this years RoboRIO. We will be giving each team at our WeekZero event the .exe of it to run.

rich-coe, 2016-02-19:
I found this FIRST posting from 12 Jan 2016 which has a link to 'FMS Off Season' a simplified version of FMS.  I'm only noting this as informational, not an endorsement.
http://www.firstinspires.org/resource-library/frc/off-season-events

AustinSchuh, 2016-09-26:
The 2016 Cheesy Arena now supports the 2016 FMS light protocol.  I'm closing this bug since this seems to be resolved by that.

closed 2016-09-26 by AustinSchuh