Team254/cheesy-arena #16: Optionally delay TBA posting for a few seconds

legoktm, 2016-08-24:
One thing I noticed at Chezy Champs and again at CalGames last year was that TBA posting was iinstantaneous, which is great, but sometimes even before the official score was announced at the venue. It definitely kills the suspense when your phone tells you the result while everyone else is eagerly waiting for the announcer to reveal it.

I would suggest putting in a delay of 10-20 seconds to account for this. I think it's more important during elimination matches though. It should probably be configurable since different announcer + scoring tea!s will take different amounts of time to reveal scores after they are finalized.

patfair, 2016-08-25:
Great idea! I'll implement it before Chezy Champs 2016.

themacman, 2016-08-25:
Just an idea: have the length able to be set in one of the configuration menus?  This makes it so events can tailor the posting  to how they announce the score.

phil-lopreiato, 2016-08-25:
Classic FMS has a separate button to post scores (show the audience display + upload data). Maybe you can do something similar, although I don't quite remember the match submit flow in Cheesy Arena.

fangeugene, 2016-08-25:
Posting to tba when the score is shown sounds like a good idea. Maybe also
auto post after a timeout in case scores don't get shown for some reason.

themacman, 2016-08-25:
@fangeugene Good idea, I think that's probably the best option.

commit 3cd216cb (2016-08-28): Move TBA match publishing from time of commit to time of showing score. Closes #16.
commit a919cb07 (2016-11-05): Revert "Move TBA match publishing from time of commit to time of showing score. Closes #16."

Fixes #36 but reopens #16.

This reverts commit 3cd216cbf417951bf3f2f5c6816186a49bc32e57.