Team254/cheesy-arena #28: Do not allow transitioning to teleop on the scoring screens before the match has started

ograff, 2016-10-09:
Scorers can accidentally transition to auto during pre-match by pressing enter, and may not notice the mistake, leading action in auto mode to be scored as in teleop mode. Lock out the scoring screens from changes during pre-match, only enable them once the match starts.

legoktm, 2016-10-10:
The only downside to this that I can think of would be not being able to train scorers on the scoring interface without actually starting a match. (You could start a test match that bypasses everyone, but that still seems non-ideal...)

ograff, 2016-10-10:
I think this is worth it. That's what we have to do to train refs on the FIRST FMS ref tablets. User error that can cause scoring issues is much worse than the slight inconvenience of running a test match to show scorers how to use the software.

legoktm, 2016-10-10:
Okay yeah, that seems reasonable.

patfair, 2016-10-16:
Compromise: transition to teleop will only be allowed pre-match if it's a
test match.

commit 093b7f64 (2016-10-16): Block autonomous commit pre-match unless it's a test match (fixes #28).
closed 2016-10-16 by patfair