Team254/cheesy-arena #29: Validate match invariants when editing the score breakdown manually

ograff, 2016-10-09:
When editing the score at the end of a match, or in match review, you can enter 1 auto crossing for position 1 and then 2 teleop crossings for position 1. This is an invalid case, and can result in a breach not being scored (I assume there is an ==2 test somewhere). In addition this gives a team more points than they should receive. Validate the <= 2 crossing per position invariant on submit, along with other invariants including but not limited to (total auto crossings + reaches) <= 3 and (scales + challenges) <= 3

patfair, 2016-10-16:
The general design philosophy behind Cheesy Arena has been to not make everything idiot-proof, because that might introduce bugs or unintentional artificial limitations, different events/users may want to do things different ways, and we can ensure competent volunteers in the necessary positions at our event. The "edit match" functionality is one place where we consciously decided not to have validation, for that reason.

closed 2016-10-16 by patfair

ograff, 2016-10-16:
I would suggest, then, a "basic" and "expert" mode, one with validation, and one without. At the very least, the issue that prompted this (3 total crossings for a defense results in it not counting as damaged, therefore not counting towards a breach) should be fixed