Team254/cheesy-arena #33: Teams unable to tether to their robot in between elimination matches

legoktm, 2016-10-10:
At CalGames 2016, we saw that teams were unable to tether to their robots between elimination matches when they just played and were going to play in the next match because they were still being controlled by the FMS. Removing them from the current match by clearing the field with "Load test match" didn't work - I had to restart cheesy-arena each time for teams to be able to tether.

ograff, 2016-10-10:
I believe if you load test match and then enter new team numbers, this should clear. That said, when using advanced network security mode, DS Control packets should only be sent out the interface attached to the DS, not over the radio

patfair, 2016-10-10:
There's a firewall rule in the AP config that is supposed to prevent this.
Not sure why it's not working.

ograff, 2016-10-10:
I have the ap config I pulled off the AP during BB if you'd like it. Perhaps that rule was removed during troubleshooting at CC?

patfair, 2016-10-10:
Could be. If you happen to have the switch config too I'd be happy to see
it; I think there's stuff there that's not in Github.

legoktm, 2016-10-10:
Hmm, we also had an issue where if teams had their DS set to an incorrect static IP, cheesy-arena wouldn't connect to them, but they could still enable/disable their robot on the field. Could that be related? (Or should I file a separate ticket for that?)

AustinSchuh, 2016-10-16:
Kiet removed the firewall at CC because it was blocking networktables and webpages teams were hosting on the roboRIO.  He attempted to block only some ports, but ran out of time (this was between matches on Friday)

patfair, 2016-10-16:
Hmm. I thought the way we had the firewall rules set up it would only
affect packets being routed between VLANs, not packets going between driver
stations and robots. It obviously wasn't blocking DS-robot UDP
communication despite there being no whitelist rule for it.

themacman, 2016-10-25:
Huh. @legoktm, if I'm correct, you are saying that every time a team tethered on the field you had to restart Cheesy Arena on the computer was running on?  I am 90% certain at CCC we were just able to load a test match and the teams were able to tether with no issues. Not sure if this has anything to do with the gear we were using, as we did have 254's whole set up from Chezy Champs.

AustinSchuh, 2016-10-26:
@themacman That's what he's saying.  It shouldn't be required to do that.  Loading a test match is a good workaround.

bsilver8192, 2017-04-18:
#47 makes it so removing a robot from the current match (I think that's the same as loading a test match), unplugging the driver's station until the fact that it has seen FMS times out (a minute or so IIRC), and then plugging it back into the same network as FMS even without any kind of firewall acts like there's no FMS. I'm not sure what exactly the behavior is expected/supposed to be, but I think it's tolerable now even without any kind of fancy firewalls.

patfair, 2017-09-02:
Loading a test match should work now to let teams tether on the field.

The firewall rules now block the FMS control packets from coming in over WiFi, but this isn't sufficient to allow tethering in all cases -- if the team is using DHCP on both devices, the DS will have a link-local address while the roboRIO will have one from the field DHCP, making them unable to communicate unless the radio is bypassed and the robot power-cycled.

closed 2017-09-02 by patfair