Team254/cheesy-arena #34: DS will link even when connected to the wrong DS

ograff, 2016-10-10:
Possibly related to #33, teams plugged into the wrong player station ethernet still had their driver station show up as green on the field monitor, and the DS did not show "Move to station X"

ThadHouse, 2016-10-12:
Related to this, a computer connected to the Cheesy Arena network with an open driver station will attempt to connect as well. This itself is probably related to the switch vlan settings however.

ThadHouse, 2016-10-12:
For the link issue, that is caused by this function.

https://github.com/Team254/cheesy-arena/blob/3c3e9ef25db91116d35a2862ba684bac57e8a0aa/arena.go#L563

which is used here

https://github.com/Team254/cheesy-arena/blob/master/driver_station_connection.go#L289

On the real field, the FMS checks the IP and VLAN of the receiving driver station, and if the team number does not match the expected team IP and the VLAN, the FMS sends a different enum which tells the driver station to move teams.

patfair, 2016-10-12:
Yeah, I saw this capability in the protocol, but I haven't figured out yet how the FIRST FMS figures out what VLAN or switch port a given connection is coming from.

ThadHouse, 2016-10-12:
My guess is that it just reads the remote IP from the TCP packet, and checks the 2 middle numbers in that against the team number. Thats how I'm doing it in my custom FMS.

commit c25a2377 (2017-09-02): Send 'move to station' signal when driver stations are plugged into the wrong spot (fixes #34).
closed 2017-09-02 by patfair