Team254/cheesy-arena #36: TBA push doesn't happen automatically

AustinSchuh, 2016-10-24:
We updated to the latest version of Chessy Arena today at the Capital City Classic, and found that the TBA push was not working.  If we hit Show Results from the main control page, and edited the URL to point to a valid match, it would push.  (I couldn't find any other code paths than that to trigger a push when reading the code)

phil-lopreiato, 2016-10-26:
@AustinSchuh is this why the playoff results didn't make it up to TBA? (I you happen to still have the data, we'd be happy to import it manually)

AustinSchuh, 2016-10-26:
@phil-lopreiato I had to leave before then.  I'll check and we'll get the data to you somehow.  Maybe I can just grab the database and trigger a push from there.

andrewnabors, 2016-10-26:
I also left before the final match. My assumption is they didn't run the "fix" anytime after quarters. Worst case is when we get the box for the Throwdown in a couple weeks we will post then. I'm not sure if anyone saved a backup off the box before storing it.

patfair, 2016-10-26:
I managed to get a copy of the DB (apparently someone set up Dropbox syncing) but I guess I'm hitting this bug because the results haven't updated yet. I don't have time to investigate right now, but here's the DB in case someone else wants to try.
[2016-capitalcityclassic.db.zip](https://github.com/Team254/cheesy-arena/files/554337/2016-capitalcityclassic.db.zip)

themacman, 2016-10-26:
@phil-lopreiato @andrewnabors Harsh (one of our other FTAs) has final scores, breakdown of teleop/auto/foul points, and whether the teams received a beach and/or capture, but not the scoring (high/low goals, crossings, etc.) breakdowns. We (he, Austin, and I) are talking over email about getting everything uploaded.

phil-lopreiato, 2016-10-27:
Awesome, thanks so much! If there's anything you need from the TBA side, let me know and I'll be happy to do it

AustinSchuh, 2016-10-27:
@phil-lopreiato I hit the magic webpage with the database provide by @patfair and it pushed.  Let me know if something else is broken.

@patfair , I'm thinking about looking at the commit logs to see when the push code moved for that page, but if you have any ideas for why it doesn't get triggered anymore except when hitting http://localhost:8080/match_play/{anynumber}/show_result , that would be awesome.

patfair, 2016-10-27:
If no-one has a copy of the DB, we should be getting the road case back
soon and should be able to grab it off of there.

Maybe I'll implement auto-backup to S3 or something in the next version.

patfair, 2016-10-27:
I managed to get a copy of the DB (apparently someone set up Dropbox
syncing) but I guess I'm hitting this bug because the results haven't
updated yet. I don't have time to investigate right now, but here's the DB
in case someone else wants to try.

patfair, 2016-10-27:
Ignore the comments I just deleted; GitHub apparently had a 20-hour delay on e-mail replies.

phil-lopreiato, 2016-10-27:
@AustinSchuh looks like everything is [all set on TBA](https://www.thebluealliance.com/event/2016cacc), thanks for taking a second look!

AustinSchuh, 2016-10-28:
@patfair 3cd216cbf417951bf3f2f5c6816186a49bc32e57 looks like it introduced the issue.  We should either roll it back, or fix it.

commit a919cb07 (2016-11-05): Revert "Move TBA match publishing from time of commit to time of showing score. Closes #16."

Fixes #36 but reopens #16.

This reverts commit 3cd216cbf417951bf3f2f5c6816186a49bc32e57.
closed 2016-11-05 by patfair