Team254/cheesy-arena #38: Cheesy Arena for Steamworks?

onenerdyguy, 2017-01-19:
Will one be made? I know, loaded question. We're hoping to use it for an unofficial week 0

patfair, 2017-01-19:
One will be made, but not before the end of the official competition season. Right now there's no time, and it's not yet clear what changes to the official rules will be desired for Chezy Champs and other events.

onenerdyguy, 2017-01-19:
Works for me. Thanks!

-Bob Henderson-
Technology Coordinator
Robotics Coach
Lake Park Audubon Schools
218.238.5914 ext 2030
www.lakeparkaudubon.com
Check us out on Facebook! <http://www.facebook.com/lpa2889>

patfair, 2017-09-04:
The 2017 version is pretty much done, and can be built from the source of the master branch. I'll wait until after Chezy Champs before releasing binaries.

closed 2017-09-04 by patfair