Team254/cheesy-arena #39: networking issues

onenerdyguy, 2017-02-03:
Apologies if not the correct place. 

I picked up the recommended equipment from the advanced networking section and am having difficulty configuring them.  I have the configs pulled, but ran into 2 issues:

1. On the switch, Vlan1 is by default administratively down. I put no shutdown into the config to bring it backup. 

2. Both radios on the Cisco AP were down, also put the shutdown in there. 

However, I'm still having an issue where I cannot connect to the AP when it is plugged into port 9 or 11 on the switch, which are still trunks. Neither the event server, in port 1, can ping/telnet into the AP, nor can the switch itself.  

I have attempted a fix by changing the BVI1 ip address to something in the 10.0.100.x/24 block, to no change. However, if I remove the ap from the switch and plug it directly into the event server, which is at 10.0.100.5, it can telnet/ping and works as expected.

Any suggestions? I'm loving cheesy arena so far, and would love to use it for our upcoming week0 event.

themacman, 2017-02-05:
Hey, don't feel bad! I'm still mastering all of the networking stuff and I set up and ran Cheesy Arena at two events last offseason.

There should be a button in the bottom right, next to the green comment button, where you can close the issue.

themacman, 2017-03-11:
Seems like I'm hitting the same issue. What did you set your gateway to? @onenerdyguy

patfair, 2017-09-02:
Closing this since there's a new switch/AP anyway.

closed 2017-09-02 by patfair