Team254/cheesy-arena #40: generated schedule does not show all matches

rich-coe, 2017-02-12:
start time: 12:00 pm
end time: 4:00 pm
cycle time 10 mins
Match count: 21   Actual end time: 4:00 pm

With 26 teams in the database, the result is
Total match count: 21
matches per team: 4
Excess matches: 3
Matches needed for +1 per team: 1

The generated schedule only shows 18 matches even though it shows that there should be 21 matches.
If you save the generated schedule, the match schedule report also only shows 18 matches.

This is with the current master version, commit a919cb077ebbad2dfebb042813685d3a9677b1c5, from 5 November 2016.

rich-coe, 2017-02-12:
The reason there are only 18 matches is because there are only 18 matches in the 26_4 template.
If the schedule says that there are excess matches, you need to adjust the cycle time or start/end time to make it so no excess matches appear.

closed 2017-02-12 by rich-coe

patfair, 2017-02-14:
Yeah, that's by design. Even if the schedule can hold more matches in the time allotted, it will truncate it such that all teams have the same number of matches (discounting surrogates).