Team254/cheesy-arena #41: Failed to configure team WiFi: Invalid WPA key

mddrouillard, 2017-02-16:
I am trying to run cheesy arena for the first time, after I have set up a Cisco Cat switch and a Cisco AP model 1131 (AIR-AP1131AG-A-K9)

The only real difference between the model you would like, and this one, is that  my unit has fastethernet port rather than a Gig port.  I have the config in the AP as I believe I should. (I think).

Anyway, after starting CheesyArea, and launching the admin page , adding a team for match play results in this on the mac ... running terminal.

/cheesy-arena.command 
goose: no migrations to run. current version: 20140823193501
2017/02/16 09:49:08 Listening for driver stations on UDP port 1160
2017/02/16 09:49:08 Listening for driver stations on TCP port 1750
2017/02/16 09:49:08 Serving HTTP requests on port 8080
2017/02/16 09:49:30 Failed to configure team WiFi: Invalid WPA key '' configured for team XXX

where XXX is the team number.

I did add the team to the list of teams, and I can log into both the switch and AP via the mac terminal command using the passwords, so I believe telnet communication is working.

I looked at the source code, and I don't see any reference to Gigethenet so I don't believe your scripts care about the AP having only fastethernet ports. I believe it is looking for the red/blue vlans and to the best of my knowledge my AP is set up the way you need it to be.

AP config is attached...

[ca-ap.txt](https://github.com/Team254/cheesy-arena/files/780578/ca-ap.txt)

Any ideas?

patfair, 2017-02-16:
It sounds just like the WPA keys weren't generated. There's a button on the team list setup page to generate them; try that? Maybe it should be automatic though...

mddrouillard, 2017-02-16:
Thanks Patrick

I did find the generate keys button. 
I am no longer getting the wpa key error.

That said, I have not been successful in getting the match start buttons to light up. Admittedly , I only put in 2 teams, so I don’t know if the software requires a full 6 teams to be “registered” before I can do an ad-hoc practice match.

My issues are likely me(user) errors in not knowing how to use the software, beyond what is documented in the guide…

md

patfair, 2017-02-16:
In order to start the match, all stations either need to have a team connected or be bypassed, which you can do by clicking the green rectangles. It's the same behavior as the official FMS, but I guess someone who's never used it before would be confused, sorry.

closed 2017-02-16 by patfair

mddrouillard, 2017-02-16:
Many thanks. 
I told you it was a user error. :)

Make sense 

Sent from my mobile so please forgive spell mis steaks.