Team254/cheesy-arena #43: Issues with 2017 driver station

rich-coe, 2017-02-18:
May not be an issue with Cheesy Arena.  
I'm writing this up in case someone else runs into same problem.

I'm testing connecting to the 2017 driver station on windows 10.
The driver station got an ip-address assigned from the DHCP server and found the robot.
When the FMS gets contact from the DS, the ip-address on the DS drops and switches to a link-local address 169.*.*.*.   Reseting the ethernet gets a new ip-address, DS attempts to reconnect to the FMS and the process repeats.

I set the DS ip-address to 10.TE.AM.X and the DS is able to connect to FMS without an issue.

andrewnabors, 2017-02-18:
It's my understanding that static IPs must still be used on the DS with Cheesy Arena. Preferably 10.TE.AM.5.

patfair, 2017-02-19:
DHCP on both the DS and RoboRIO should work from 2016 onwards. I don't think I've seen this particular issue before, though. There was an issue with DHCP on the robot not working with certain versions of the AP firmware, but this doesn't sound like the same thing.

Setting a static IP on the DS should work fine as a fallback option.

rich-coe, 2017-02-19:
DHCP worked just fine. Just after the DS connected to FMS, the laptop dropped the ethernet connection and switched to a link local address.  Which is why I think the DS software is causing/doing it.  I'll try tracing the network.  I'll have more time to investigate it after tomorrow's big day.
I'm just glad I caught it in testing.

themacman, 2017-02-22:
We experienced similar issues on occasion this offseason. Setting the static IP to 10.TE.AM.5 and the subnet mask to 255.255.255.0 like Nabors said fixed it. 

I'm setting up some hardware this week so I'll try and mess around to reproduce the issue if I can.

ejordan376, 2017-05-07:
That was in issue with the kick-off version of driver station (17.a11).

patfair, 2017-09-02:
I've been testing for Chezy Champs, and this doesn't seem to be a problem any more.

closed 2017-09-02 by patfair