Team254/cheesy-arena #44: "No valid version found"

JamieSinn, 2017-03-30:


rich-coe, 2017-03-31:
The message is coming from the cheesy-arena program.
Searching the dependencies, it is coming from goose.
I suspect you may an event.db in the current directory that cheesy-arena cannot read.
If this is the first time running the program, I'd remove it.  Otherwise, rename the file and try again.

JamieSinn, 2017-03-31:
There is no event.db file in the current dir. Running CMD as admin gives the same error. And yes, this is the first time running it.

themacman, 2017-03-31:
I'm hitting that on Linux (Ubuntu 16.04 LTS) too. I've gotten around it by cd'ing to the directory where the executable is and then putting the entire thing inside apostrophes, but I haven't been able to get much further (I straight up can't get the switch and AP to talk to the computer no matter what). Maybe this issue and that are related?

patfair, 2017-03-31:
Are you running cheesy-arena.exe from the same directory it was compiled into? There's a runtime dependency on the db/migrations directory to define the DB structure the first time it is created.

JamieSinn, 2017-03-31:
Ah, there we go. 
That would be the problem. Thanks Pat.

Would be a good addition to the readme.

closed 2017-03-31 by JamieSinn