Team254/cheesy-arena #45: Layer 2 Switch + Router instead of 3550

JamieSinn, 2017-04-01:
I have a 3524 from my old CCNA days, but no 3550. 

I do however, have 3 2821 routers. Does cheesy arena require the line speed routing from the L3 switch? Or can I just modify the config and have it route over to the 2821?

JamieSinn, 2017-04-01:
I'll probably pick up a 3550 sometime, but in the meantime would this be a feasible idea?

themacman, 2017-04-02:
Isn't this going to require changes to how Cheesy Arena itself configures the VLANs over telnet in addition to the config file changes?

JamieSinn, 2017-04-02:
Since this is simply for testing, I won't need to have it configure anything, I'll just have it set static for my team to test.

themacman, 2017-04-02:
If you're configuring VLANs manually and changing configs I don't see why it shouldn't work for what you need. And if it doesn't I have faith you'll figure it out :P

patfair, 2017-04-02:
Should work fine as far as I can tell. The configuration that Cheesy Arena does is pretty vanilla, and the bandwidth consumed by robots is not that much.

JamieSinn, 2017-04-03:
Cool thanks, that's what I was looking for.

closed 2017-04-03 by JamieSinn