Team254/cheesy-arena #49: Linksys Access Point Issues

themacman, 2017-10-09:
Opening an issue here to document problems in relation to the new Linksys field access point used in Cheesy Arena this season. All these seem to be intermittent issues.

Issue 1: Team SSIDs will not update on the AP. This is most commonly seen with the AP retaining the team numbers from the previous match. The temporary fix for this is either reloading the match (sometimes works), stopping and restarting the instance of Cheesy Arena on the arena computer (nearly always works), or in severe cases, restarting the access point.

Issue 2: Access point experiences intermittent crashes. At Chezy Champs the AP would crash and restart (somewhat randomly), usually during the configuration of the networking hardware and once during a match. I attempted to diagnose during one of the breaks, but with limited time it seemed that OpenWRT doesn't retain logs after restart (although I could definitely be missing something).

There were additional issues at Beach Blitz this weekend with the AP, and although I helped remotely diagnose some of them I can't speak to exact details at the event. Maybe @ThadHouse could as I know he was there.

legoktm, 2017-10-17:
At CalGames we ran into issue 1 a lot (every 4 matches on average), and restarting the arena always fixed it. We think we ran into issue 2 once, and after restarting the AP twice, and then basically everything on the scoring table (switch, PLC, AP, cheesy arena) all at the same time, it went away.

patfair, 2017-10-26:
I have another access point at home, but I was unable to reproduce this issue with it. Maybe it only shows up under a certain amount of connection traffic from robot radios.

I added a timeout to the running of the SSH command in 470407b801a399e4281d41d38d9c138bde26f697, which will maybe make it unnecessary to restart Cheesy Arena if Issue 1 happens. It'll log an error message if that happens; let me know if it shows up during CCC or MTTD.

themacman, 2017-10-30:
Update: we got the  SSH time out at least 15-20 distinct times at CCC. There was also one instance where we got a "TCP dial timed out" or somesuch similar error and the AP got stuck in a reboot loop and had to be restarted.

patfair, 2017-10-30:
Did restarting Cheesy Arena resolve the SSH timeout problem, or would just reloading the match have fixed it? Any thoughts on what conditions help reproduce the issue?

themacman, 2017-10-30:
Reloading the match _usually_ fixed it, and sometimes the workflow was loading a test match and then reloading the match. Usually when it happened we'd still have robots linked to the AP, but that shouldn't matter. On reproducing it, I have no clue at all. Maybe try linking multiple robots to the AP and loading the match then? The issue seems to only show up under robot traffic.

patfair, 2017-10-31:
If just reloading can fix it, I'll try adding some retries when we detect a timeout.

AustinInSeattle, 2017-10-31:
For issue 1: https://godoc.org/golang.org/x/crypto/ssh#Dial
The Dial method you're calling might not be the right one of the two, as the standard does actually call SSH by default, and handles the config and everything. Maybe the differences between the two can cause issues?

For issue 2: OpenWRT does not store logs, but this can be changed... (Look up the docs for the Radios.)

themacman, 2018-08-27:
Closing as per d161a6a115b5bd5566db0b74eddb02800e2a61a8, hopefully that does it!

closed 2018-08-27 by themacman