Team254/cheesy-arena #50: Allow customizing WiFi channels on a per-team basis

legoktm, 2017-10-17:
Right now the wifi channel is hardcoded to 157 in `templates/access_point.cfg`. It would be nice if it was easier to customize this (changeable via web UI) and could be set on a per team basis so we can spread the networks across different channels.

At CalGames we were unlucky and the venue already had a few networks on 157 :/

themacman, 2017-10-18:
Agreed, however I don't really think we need per-team adjustability. I'm fairly certain the real field switches over all robot networks when FRC Engineering changes the channel, so that is sufficient IMO.

patfair, 2017-10-18:
I'm not sure I fully understand the "set on a per team basis" ask, but I don't think that's possible. In reality there is only one network on one channel, and the six team SSIDs are multiplexed within that.

That said, we can definitely make that channel configurable in the settings instead of hardcoding it in the template.

legoktm, 2017-10-20:
It's probable that I misunderstood exactly what our FTA was asking for at that time, so what you've described seems good enough.

commit 11d4f4a8 (2017-10-25): Make WiFi channels configurable (fixes #50).
closed 2017-10-25 by patfair