Team254/cheesy-arena #60: Can you release a Windows version too this year?

YeveRivkin, 2018-05-17:


ebot1234, 2018-05-17:
You know you can compile all the source code pretty easily for windows. All the info on how to do that I'm pretty sure is in the wiki.
-Ethen

patfair, 2018-05-17:
I can't release a Windows version since I don't own a Windows computer, but if someone else can help me out I'll gladly upload the zip to the release.

ebot1234, 2018-05-18:
I can do it if you want, I have previously compiled the code for windows.
-Ethen
ebot14rocks@gmail.com

patfair, 2018-05-18:
Awesome, that would be great. Probably won't be releasing anything until September, though.

ebot1234, 2018-05-18:
Ok, cool.

patfair, 2018-10-07:
I just uploaded a release for macOS, so if someone wants to create a Windows zip now I'd be happy to add it.

I tried out Go's nifty cross-compilation capability, but unfortunately the sqlite driver I'm using has dependencies on C code that is much harder to cross-compile.

themacman, 2018-10-08:
Cool, I'll see about doing a Windows build sometime this week. If someone has more downtime than me, feel free to beat me to it.

themacman, 2018-10-31:
Can we close?

closed 2018-11-03 by patfair