Team254/cheesy-arena #63: FTA Display

ejordan376, 2018-09-29:
Can you bring back the FTA display from the 2017.1.2 version, also keep the existing field monitor.

patfair, 2018-10-05:
Yeah, I'll get something in place by Beach Blitz.

FletcherS7, 2018-10-07:
looks good.
@ejordan376 Can you add the fuchsia color we tested at CalGames to the display?

FletcherS7, 2018-10-15:
@patfair Would it be possible to sill restore the 2017 FTA display?

patfair, 2018-10-16:
Is there any particular feature you want from the previous FTA display that is missing from the field monitor? Next year I'm planning to add an additional page with more textual data, like team notes and event logs, so I'd prefer to avoid ending up with three different displays.

FletcherS7, 2018-10-29:
low battery voltage level in pre match state should be anything below 12v
low battery voltage level during match play should be 6v and below

themacman, 2018-10-31:
The following may be helpful to when figuring out what you want/don't want to implement:

[FIRST FMS Match Log Interface](http://wpilib.screenstepslive.com/s/fms/m/fieldmonitor/l/591663-viewing-match-logs)

[FIRST FMS Field Monitor](http://wpilib.screenstepslive.com/s/fms/m/fieldmonitor/l/592785-live-monitor)

Screenshot from FIRST's FTA Notepad:
[image]

commit 30b0679b (2020-04-01): Add ethernet connected, trip time, missed packets, and tooltips to the field monitor (closes #63).
closed 2020-04-01 by patfair