Team254/cheesy-arena #66: Ref Panel can not force light changes.

FletcherS7, 2018-10-07:
When selecting the purple or green lights on the head ref screen the lights are sometimes not triggered, Probably something to do with field status? Can an option to force the lights to change be added? But to only force the change if a match is not active.

patfair, 2018-10-07:
Right now the referee can only set the lights while the arena is in post-match state. Can you clarify whether this issue is that isn't working as intended, or that you want to add a use case (i.e. what are the other times besides post-match when the referee should have control over the lights)?

FletcherS7, 2018-10-07:
When the field is E-Stopped. During "Pre-Start".

patfair, 2018-10-07:
When the field is E-stopped it goes to post-match state, so the referee should already have control. What's the use case for controlling the lights pre-match?

FletcherS7, 2018-10-07:
The ref we had at CalGames always triggered the purple lights for volunteers then finalized scores before triggering the green lights. For some reason the lights would either not switch to green or switch to green for less then a second before switching to "Not Owned".

And after rebooting Cheesy-Arena after it crashed (and triggered an E-Stop) the ref panel light buttons would not function.

patfair, 2018-10-07:
Got it. There's not much we can do to intelligently recover from a crash, since all state is lost. In that case it's probably best for the scorekeeper to just manually override the lights.

We can do some stuff though to make sure the lights go on for a minimum amount of time after the score is committed.

FletcherS7, 2018-10-07:
Adding a minimum time the lights are on for sounds good (as long as the ref panel overrides the minimum). 
Would it be possible to add a force green button to the ref panel that does the same as the button on the PLC testing page?

FletcherS7, 2018-10-29:
Another idea
Have the Scale lights remain green till all robots are fully connected "green field"

themacman, 2018-10-31:
You can't force cleanup/reset lights from the head referee tablet using FIRST FMS unless you are in a post-match state anyways. However, the scorekeeper can force these lights on from the "Options" tab in the Match Play interface irrespective of the match state. This seems like a good circumstance to follow the FIRST case and do it that way. Image below of options.

[image]

And re:calgames, the referee should probably follow the FIRST flow of submitting scores before signaling cleanup, making a special case in cheesy arena for something one referee does isn't the most reasonable in my opinion, just my 2 cents.

FletcherS7, 2018-10-31:
The reason for this was the ref wanted to force the lights green after cheesy-arena crashed and was restarted. 
but cheesy-arena would not allow the ref to set the lights green till after a match

themacman, 2018-10-31:
Understood. I am saying at an event using FIRST FMS, the head referee would turn around and tell the scorekeeper to do it, because the head referee does not have that capability. This is one instance where I strongly believe parity should be kept with FIRST. I don't think this is something that needs to be implemented on the referee side, there are many other development tasks that should be prioritized over this.

FletcherS7, 2018-10-31:
I am renaming this issues as the suggestion has changed

FletcherS7, 2018-10-31:
Revised comment
Have the scale lights (or in the future some of the field lights) remain green till all robots are full connected. 

As a problem we encountered over the off-season events was teams not seeing the lights green since they do not remain green. 

It would also be useful to add Force Purple and Force Green to the "Match Play" screen in cheesy-arena

Ewpratten, 2018-11-11:
@themacman where did you find this app?

FletcherS7, 2018-11-12:
solved in 42a6041ec6c1e86ca7682208450e626742341d95

closed 2018-11-12 by FletcherS7

themacman, 2018-11-16:
@Ewpratten that is a screenshot from the official FIRST FMS, which is not publicly available.

Ewpratten, 2018-11-16:
Oh. Ok