Team254/cheesy-arena #70: Question about deiveratations

Ewpratten, 2018-10-19:
Due to the fact that my team does not have access to the hardware required for testing, I figured i should ask here.

When a driverstation is connected to a FMS, is it still able to connect to and control a robot over it's own wireless interface? Or does the robot have to also be connected to FMS?

FletcherS7, 2018-10-19:
When you configure the robot radio at competition it is put in Bridge Mode. 
The radio is then only able to connect to the Field Access Point. 

So the answer is no.

Ewpratten, 2018-10-19:
If I didn't reconfigure at a competition (ignoring rules for now) would it
be possible?

patfair, 2018-10-19:
It probably wouldn't work -- the driver station laptop would be assigned IP addresses in the same 10.TE.AM.x subnet on both its wireless and wired interfaces, so the operating system would be confused as to how to correctly route packets.

Ewpratten, 2018-10-19:
Alright. Thanks for the help!

closed 2018-10-19 by patfair

terbos90803, 2018-10-19:
We typically have to static IP the driver station on the field network.  How critical is the 10.x.x.5 for the DS IP?  I seem to recall that it's not critical, at least on the robot side.  So maybe a dual network setup might work if the WiFi side is an available address other than .5.  It's at least worth an experiment.

FletcherS7, 2018-10-19:
Why do you want to connect to the robot via a wireless adapter during a completion?

terbos90803, 2018-10-19:
The OP was looking to test CA at home, apparently using a setup without a router/AP.  Their question was whether it was possible to wire the laptop to another computer running CA/FMS and use the laptop's builtin WiFi to connect to the robot.

I think this should be possible, or at least is worth an experiment.

ejordan376, 2018-10-19:
In that case, I think you could have the second laptop connect to the robot radio.

terbos90803, 2018-10-19:
That should work too.  Might be easier as well since the robot radio is known to route packets correctly, whereas bridging network adapters on Windows won't necessarily route.

Ewpratten, 2018-10-19:
So I am not actually looking to test CA itself, but a few of us at 5024 are
building our own FMS "clone" that was inspired by this repo. It runs on a
raspberry Pi and uses an Arduino to replace all the expensive field tech.

Currently, we are using a client that runs alongside the driverstation that
injects keystrokes into driverstation to enable and disable.

The driverstation laptop is wired into a switch that connects to the
reapberry pi and also uses wifi to talk to the robot.

I would like to replace this hacked together solution with one that uses
the actual FMS protocol, but I can't do that until I find out if
driverstation would allow wifi to the robot and ethernet to FMS at the same
time.

That is the source of this question.

MrLtMajor, 2018-10-19:
This does not sound safe.

Ewpratten, 2018-10-19:
@mrltmajor why do you say that?

FletcherS7, 2018-10-31:
I would highly suggest you do not inject keystrokes and use the drive station software in this way.

Ewpratten, 2018-10-31:
We have now moved on to an alternate method. Thank you for the advice.