Team254/cheesy-arena #71: Access point buffer overflow

FletcherS7, 2018-10-29:
b35a70d added the feature for cheesy arena to retry setting the access point config till it worked. 

unfortunately this seems to cause the access point to stall and be unable to accept new configs. 
Rebooting cheesy arena seemed to fix this issue.
maybe instead add the option to reload the match. (right now this is done by loading a test match then loading the match back up, but b35a70d causes problems requiring rebooting of cheesy arena)

FletcherS7, 2018-10-29:
@ejordan376 can explain in more detail

ejordan376, 2018-10-29:
During quals and elims we must of restarted CheesyArena 30-40 times. The ap config would not change. There was no log messages on the console that CheesyArena was even trying to configure the AP. Restarting the AP made no change. Loading a test match then reloading the qual match made no difference. Sometimes it would be fine for 6-7 matches other times it would fail every match.

patfair, 2018-11-03:
Hmm. Seems like there is a failure mode of the AP that we still don't understand, and that isn't reproducible without real robot traffic. I'll try to debug more next time I have a robot available to test with.

FletcherS7, 2018-11-03:
Till then could you revert that change?

ejordan376, 2018-11-10:
We have been trying to debug the AP here at MTTD. We have noticed the the AP crashes of the first try of pushing the config, and this causes Cheesy Arena to hang up in a state is does not auto retry. While in practice, when it gets in this state the change requests will back up in the AP Buffer until it overflows, and then it will print to the log "Failed to configure team WiFi:WiFi config request buffer full"

patfair, 2018-11-11:
Thanks for the info! So it looks like the SSH command just hangs indefinitely when the AP crashes, which is why it doesn't do any further configuration attempts until you kill and restart it? And that thread isn't crashing out or producing any further output, right?

If so, I think I should be able to fix this by adding a timeout to the SSH command that just continues the AP loop.

ejordan376, 2018-11-11:
The ssh is hanging. No errors. I just sent you a screen shot.

patfair, 2018-11-11:
I just added a timeout in dc74b1dc19e3a98fff468f9043888eb0da059619. If you want, you can pull it down and try it out this afternoon.

ejordan376, 2018-11-11:
It works. We just had the AP crash, and cheesy arena was able to auto recover.

patfair, 2018-11-11:
Awesome! Thanks for the help debugging.

closed 2018-11-11 by patfair