Team254/cheesy-arena #83: ***FEATURE SUGGESTION*** Simulated PLC using Raspberry Pi or Arduino Board

770352, 2019-01-21:
The possibility to use a raspberry pi's GPIO header or an Arduino Board as a simulated PLC would be very nice. This would allow teams to use in-expensive electronics to have better a better FMS during offseason events. (Mainly, Having actual E-Stop buttons, Automatic Scoring, and outputs for devices such as boilers or sandstorms.)

ebot1234, 2019-01-21:
@770352 Look at this repo. It is still under development but I created it for this issue. It will have most of the Modbus/PLC inputs and outputs via multiple arduinos for the field I/O depending on what Cheesy Arena supports. 

https://github.com/ebot1234/PLC-Sim-Arduino

ebot1234, 2019-01-23:
Yes you could theoretically run an Arduino sketch/program on a Raspberry Pi. Here is a link how to simulate the Arduino framework on an R Pi. As for the GPIO header pins being simulate I don't know if that will work with the simulate Arduino framework. 

https://www.deviceplus.com/how-tos/raspberrypi-guide/how-to-run-arduino-sketches-on-raspberry-pi/

ejordan376, 2019-02-03:
Raspberry Pi could work for E-stops but things like the vaults last year used AB IOLink laser distance sensors that would not work with a Pi. Also Allen Bradly does lots of donations to schools, a local rep could get you an L18 and a few armor-blocks fairly easily.

770352, 2019-02-04:
I was going for a cheap FMS that anyone can obtain parts for, Either using
a Raspberry Pi or Arduino. I was hoping for a Raspberry Pi solution because
I only have raspberry pis.

ebot1234, 2019-02-13:
Here are the repositories for the arduino field I/O stuff.
Modbus Server:
[https://github.com/ebot1234/Python-Modbus-Server.git](url)
Arduino Server between Modbus Server and Arduino
[https://github.com/ebot1234/Arduino-Comms-Server.git](url)
Arduino Programs:
[https://github.com/ebot1234/Arduino-Field-I-O.git](url)

There is still tons of work to be done but I can read and write values to CA right now just from the Modbus Server. I still have to work on the arduino server and arduino programs

ebot1234, 2020-01-27:
@770352 If you are still interested in this feature, you should look into OpenPLC, which can run on either a RPI or computer and it simulates the features of a PLC. For remote IO like the AB ArmorBlocks you can use an Arduino or probably a RPI using Modbus TCP/IP.

https://www.openplcproject.com/

cpapplefamily, 2021-04-25:
@ebot1234 are you still developing this?  A few weekends ago I used a RaspberryPi solution using websocketapp to send key strokes to CA like the manual scoring pages do. I like this option because it did not disable manual scoring like when enabing the plc. I got the basics of the solution from the Isreal group who used it at their event. Some edits to the score template where made to disable scoring from this input when the match is not in Auto or Teliop. 

I'm now just coming across this and debating the next step. Keep developing on the websockets or the simulated plc. Maybe CA needs to have on the setting page in addition to the enabing plc an option to enable microcontroller

ebot1234, 2021-04-25:
@cpapplefamily I can come back to it, I haven't really been working on any coding projects lately but I'd be glad to get back into it. I was thinking that maybe a socket or something could be opened on the CA side and allow UDP comms to an RPI and send a string with info regarding the inputs and outputs. Something was used similarly with Automation Directs PLC for an offseason in 2014. So I think we could whip something up if you wanted to help out.

TechplexEngineer, 2021-04-26:
Folks in this thread may be interested in looking at JMS https://github.com/JaciBrunning/JMS

Which describes itself thus:
> Another alternative Field Management System for the FIRST Robotics Competition.
>
> JMS is based heavily off the official FRC FMS, as well as Cheesy-Arena. JMS differs in that the bulk of the network operations are performed onboard the JMS server itself, requiring minimal external and increasing the reliability of the system.

ebot1234, 2021-04-30:
So one of my concerns right now is that some Rockwell sensors such as the
IO-Link components wouldn't work with the GPIO of the RPis. It could be
possible to interface a Modbus TCP linked IO-Link Master to the RPi but the
cost for those are upwards of 400 so that's just something to consider.

cpapplefamily, 2021-04-30:
The idea i view with using an alternative micro controller or CPU is to simplify all the hardware. Currently sure I'm focused on discrete digital inputs for game peice counters and there see other configurations in not considering. 

How much of the io-link data is really being used. In my experience an io-link device can be used as a discrete or analog sensor. The io-link devices shine with plug and play where the PLC automatically configures the device and the sensor health where that data can be used for preventing failure. There is usually a sensor option that is not io-link enabled to sense the input required. 

I suspect many off season or week zero type events could benefit with a more obtainable option.

ebot1234, 2021-04-30:
Oh okay...I was not aware of the ability to just use the IO-Link sensors as
a normal analog or discrete sensor. Since that is the case then this option
would work quite well. As to my knowledge, I don't think they use that much
of the IO-Link capabilities. I know for CA the only data say for this year
was the colors and intensity being reported to the PLC.

cpapplefamily, 2021-04-30:
I recall skimming by that somewhere but skipped because at the time I was focused on power Cell counters. 

Did it say what sensor was used?

ebot1234, 2021-04-30:
AB 46CLR IO-Link Sensor I think was used for control panel stuff

cpapplefamily, 2021-04-30:
> AB 46CLR IO-Link Sensor I think was used for control panel stuff
> […](#)
> -- Ethen Brandenburg, FIRST Volunteer, National Instruments roboRIO Robot Control Badge Holder

Fun looking sensor. Looking through the documentation it seems there are 3 discreet outputs that can be programed to up to seven colors. 

`The ColorSight sensor also can be programmed to detect and store 
multiple colors in up to seven different color channels with the help of 
the local teach button. Upon recognition of the taught colors, up to 
three discrete outputs can be activated locally in standard IO mode. An 
additional four virtual outputs can be configured when connected using 
IO-Link.`

cpapplefamily, 2021-04-30:
If I where to be building a Offseason event system I would likely attempt to use the Rev Robotics sensor we all received in 2020

ebot1234, 2021-04-30:
Yeah, i've been thinking about using a different sensor and frankly, a few
zip ties to the crossbar on the control panel would suffice for a small
offseason to secure the sensor.