Team254/cheesy-arena #87: Issue with blue lights/E-stops

Techno11, 2019-08-05:
I'm not 100% on the main stack, but it is not currently working in my setup. I plan on debugging that later. 

Estops:
[In the Wiki](https://github.com/Team254/cheesy-arena/wiki/Default-IP-Addresses-and-Credentials), it says the Blue CommandBlock should be @ 10.0.100.42, and the red @ 10.0.100.41.  When configured in this way, the blue lights flash fine, but the E-stops trigger the RED e-stops on the PLC/LED testing page.  To make the RED E-stops work for the RED Alliance stations, I had to reverse the IP addresses, so RED was on 42 and Blue was on 41.  Is this a bug on your guys' end too? Or have I mis-wired something?  The Blue Alliance E-stops on the match play and test are also locked into the E-Stop, and as such I can't run a test match. (But the E-Stops are released and the orange alliance e-stop lights are not on)

--Lights--
When configured in the reverse-IP-config mentioned earlier, only one of the blue alliance lights flashes. (The bottom of the three on the control-block)  The other two are solid. However, the e-stop light works for all three, but it does not transfer to the software.  All three red stack lights seem to work fine. (E-stop lights included).  

I am using the most recent commit.  I will try Last Year's know good release and update this issue with what I find.  

Thanks in advance for any/all support!

Techno11, 2019-08-05:
Last year's release does not work. Is it possible that because the Field E-stop is not wired in that more issues would be caused?

ejordan376, 2019-08-05:
did you get the plc code from me?

Techno11, 2019-08-05:
No. Where can I find that?

ejordan376, 2019-08-05:
https://github.com/ejordan376/Cheesy-PLC

one day I'll post a write up on it.

Techno11, 2019-08-05:
Out of curiosity, will I need Rockwell's Software to flash this to my PLC? It is a CompactLogix L24ER and has both a USB port (which I have used with RSLinx) and what I can assume is an SD Card slot.  If not, what software also works?

ejordan376, 2019-08-05:
You will need version 20 of RSlogix 5000. There is a way to do it with the sd card, I'm not sure how to do it.

Techno11, 2019-08-05:
Thanks. For anyone who may stumble upon this issue, and doesn't have access to the RSlogix Software, [this video](https://www.youtube.com/watch?v=na5QcRh3aVg) let me program the plc. I ran out of time to test fully today, and will try some more tommorow

Techno11, 2019-08-06:
Would you be able to provide an XML of the plc software? I cannot seem to load the ACD file from the SD Card. According to the manual, it would need to be an XML file.  (I don't have access to the RSLogix software)  Thanks!

ejordan376, 2019-08-07:
[CurrentApp.zip](https://github.com/Team254/cheesy-arena/files/3474897/CurrentApp.zip)

Techno11, 2019-08-07:
What should the ArmorBlock ip addresses be? I'm not getting any active connections according to the PLC web page.

ejordan376, 2019-08-07:
PLC | Scoring table cabinet | 10.0.100.40 
Red PLC ArmorBlock | Red alliance station cabinet | 10.0.100.41    
Blue PLC ArmorBlock | Blue alliance station cabinet | 10.0.100.42    

Red network switch | Red alliance station cabinet | 10.0.100.45
Blue network switch | Blue alliance station cabinet | 10.0.100.46

Techno11, 2019-08-08:
Fixed

closed 2019-08-08 by Techno11