Team254/cheesy-arena #88: No Robot Voltage on Control Page, Driver Station Not seeing Robot

Techno11, 2019-08-08:
Not seeing voltage from robot on Match control page and while the DS sees FMS, it does not ever connect to the robot.  The only error I'm getting in the console is that it can't connect to the 10.0.100.2 switch. Why does it need to connect to the switch?  Is that what is limiting us from seeing the robot voltage?

Techno11, 2019-08-08:
I feel stupid with my original comment, realizing after looking at the switch.go why the switch needed configuring.  Now, having changed the switch.go to work with the cisco switch that we have, I cannot get the app to connect to the switch. I keep getting the error
``Failed to configure team Ethernet: read tcp 10.0.100.5:53347->10.0.100.2:23: wsarecv: An existing connection was forcibly closed by the remote host.``  When looking at the switch.go, I don't see any thing that uses a username, but when I telnet into the switch, a username is required.  I'm still investigating, and may need to change some more commands.  Just wondering if I'm overlooking something.

patfair, 2019-08-09:
I don't think you're overlooking anything; my switch doesn't prompt for a username. Perhaps you have a slightly different hardware model or firmware version? Cheesy Arena was written against a WS-3560G-48TS-E with IOS 12.2.

FletcherS7, 2019-08-09:
I would suggest wiping the switch 
Command: write erase

Then copying the config to the switch from TFTP
command: copy tftp start

Then reload it.

If that fails. Try configuring the switch by manually entering each config command.
Your switch must support layer 4 inter vlan routing.

Techno11, 2019-08-09:
Yeah, the switch that was purchased to go with our field is an SG300 which has been a pain in the rear for far to long.  Had trouble configuring it from day one, all the commands for it are different than the Catalyst line, and a lot of the features are just simply not there. The telnet from the app won't work because of the funky username/pw requirement, the SSH is also wonky because of the un/pw requirement. Just ordered 3 48 port catlysts 3500 series from ebay for less than a quarter of the price we paid for one 10 port sg300. Sometimes, you just gotta go with what works.  Hopefully from there it will be a breeze.

Techno11, 2019-08-15:
Got the new switch today, works like a charm. Thanks guys! Ran a successful match through cheesy with robots.

closed 2019-08-15 by Techno11