Team254/cheesy-arena #94: Add option to Reload match

FletcherS7, 2019-09-29:
Cheesy Arena does pre starting when the match is loaded unlike FIRST FMS which has pre start as a separate step. Add ability to Reload a match (the current work around is to load a test match then load the match again)

patfair, 2019-09-30:
Which particular problem are you trying to fix by reloading the match? Depending on what it is, we may be able to fix it either by just clicking the "Load" button for that match again, or by adding an internal feedback loop.

FletcherS7, 2019-09-30:
It's not a particular problem. It's best described a network initialization problem.
This is a problem found with FIRST FMS as well. 
**I** think it's related to the radios, but who knows, "Pre Starting" fixes it.

FletcherS7, 2021-11-01:
Still encountering cases where we need to load a test match, then load the current match again.
Low priority on this request, But I still think a reload match option would be useful.

FletcherS7, 2023-10-26:
@patfair 
Currently clicking the load match button again does not re initialization the match and the field network. Would it be possible to change the behavior so it does? or add another button that does that.

Still running into problems that require loading a test match and then loading into the match again.

commit 888b8d46 (2023-10-29): Change match load to always redo AP and switch configs even if teams are the same (closes #94).
closed 2023-10-29 by patfair