Team254/cheesy-arena #106: Unable to build due to unfindable module

770352, 2021-03-15:
I have been unable to build cheesy-arena due to httpauth-go not existing anymore.
Here's the error
```github.com/Team254/cheesy-arena imports
        github.com/Team254/cheesy-arena/web imports
        bitbucket.org/rj/httpauth-go: cannot find module providing package bitbucket.org/rj/httpauth-go

C:\Users\Administrator.WAVELENGTH\Desktop\cheesy-arena-lite>go get  bitbucket.org/rj/httpauth-go
go get bitbucket.org/rj/httpauth-go: reading https://api.bitbucket.org/2.0/repositories/rj/httpauth-go?fields=scm: 404 Not Found

C:\Users\Administrator.WAVELENGTH\Desktop\cheesy-arena-lite>go get github.com/Team254/cheesy-arena
github.com/Team254/cheesy-arena imports
        github.com/Team254/cheesy-arena/web imports
        bitbucket.org/rj/httpauth-go: cannot find module providing package bitbucket.org/rj/httpauth-go```

ebot1234, 2021-03-15:
What version of GO are you using? I had this problem when I used the latest version but after I switched to 14.x I was able to get everything correctly

patfair, 2021-03-15:
Try doing `go get github.com/Team254/cheesy-arena@HEAD`. Cheesy Arena doesn't depend on that `httpauth-go` library any more, but by default Go's new behavior fetches the latest tagged version (which is from 2018) instead of the latest committed version.

770352, 2021-03-15:
Was able to get it built, However I have a new issue. Running on Windows 10, I encounter an error when running the exe file.
```C:\Users\Administrator.WAVELENGTH\go\bin>cheesy-arena.exe
2021/03/15 18:45:04 Error during startup:  no valid version found```

patfair, 2021-03-15:
Ah, right -- that's because it's looking for the other files besides the executable binary, and the new module method doesn't put them in the expected place. Sorry for the misdirection.

Instead, try fetching with `GO111MODULE=off go get github.com/Team254/cheesy-arena` which should use the old behavior.

770352, 2021-03-15:
"GO111MODULE is not recognized as an internal or external command, operable program or batch file."

patfair, 2021-03-15:
Hmm, that's got to be a problem specific to Windows. `GO111MODULE=off` sets an environment variable on Linux or macOS.

I can't test it, but I think the Windows equivalent is `SET GO111MODULE=off`.

770352, 2021-03-15:
Still no dice, Same version error, Just a different time stamp

closed 2022-04-30 by patfair