Team254/cheesy-arena #109: Unable to updates rankings to TBA

GuyNoimark, 2021-04-03:
Hi, last week we held an off-season competition and for some reason I can't get the rankings updates on the blue alliance.
Any Idea why? Regular matches and results have updated with no problem.

[https://www.thebluealliance.com/event/2021isoir2](https://www.thebluealliance.com/event/2021isoir2)

patfair, 2021-04-04:
@GuyNoimark thanks for the report; I'm working on reproducing it. Would you be able to share your `event.db` to help me test? (no worries if not)

GuyNoimark, 2021-04-04:
Yes, no problem :)

[Orbit_Infinite_Restart-20210325192640 - 2021isoir2 (1).zip](https://github.com/Team254/cheesy-arena/files/6254080/Orbit_Infinite_Restart-20210325192640.-.2021isoir2.1.zip)

Also Just mentions that during the event the TBA API wasn't updated to 2021 so we got this error code:

> 2021/03/25 12:28:09 Failed to publish matches: Got status code 400 from TBA: {"Error": "Valid score bre

akdowns for 2021 are: set([])"}

with that said I contacted the TBA and they updated the API. But the TBA Event Wizard isn't excepting an `event.db` file so I needed to find a solution. What I did, at last, was to restart cheesy arena (a couple of days later), upload the old `event.db`, And then edit a match and save it so the cheesy arena will push the data to the TBA. 

So, Maybe add a button to force push the data to the TBA? Also, in the reports tab, I will suggest adding an FMS-like report that the TBA Event Wizard can accept.

patfair, 2022-04-30:
This was eventually fixed before Chezy Champs 2021; closing the issue.

closed 2022-04-30 by patfair