Team254/cheesy-arena #111: WRT1900ACS will not configure after removing second AP

cpapplefamily, 2021-07-08:
I have been testing a Cheesy Arena system using two Linksys AP set to 10.0.100.3 and 10.0.100.4. I removed the second AP from the setup page and saved the Page.  Now as I attempt to load matches the first three team get configured but the others stay set to the default no-team-4, no-team-6, no-team-6

I have tried to restart Cheesy Arena with these new setting but continue to get
2021/07/08 12:42:00 Error configuring WiFi: dial tcp :22: i/o timeout
2021/07/08 12:42:00 WiFi configuration still incorrect after 72 attempts; trying again.

How do I get the system to return to use just one AP?

cpapplefamily, 2021-07-08:
I ended Up deleting the Database and it now works.

patfair, 2021-07-08:
The specific setting that controls whether the second AP is enabled is "Second AP Team Channel (5GHz)" -- it needs to be set to "Disabled" to go back to using just one AP.

cpapplefamily, 2021-07-08:
> The specific setting that controls whether the second AP is enabled is "Second AP Team Channel (5GHz)" -- it needs to be set to "Disabled" to go back to using just one AP.

Ok seems that adding a Second AP everything just seems to move along with out rebooting. I did try and set the value to "Disabled" with not luck but did not restart the application. A restart after disabling the second AP did seem to work. 

What I did notice before rebooting the application though is the WiFi Overview in the AP did show all 6 teams after reset to single AP. The CA counsel out put and team status area was indicating error. 

In the end it was a poorly timed reboot on me.

cpapplefamily, 2022-01-16:
Removing the frequency and unselecte

closed 2022-01-16 by cpapplefamily