Team254/cheesy-arena #115: CC2021 Enhancement Requests

neyre, 2021-11-01:
A few nice to haves after Chezy. Fully understood that you won't do some or all of it, but wanted to document:

- PDF report with elims alliances (including team names, locations, and how far the alliance made (eliminated in QF1, still playing, etc).
- Auto-refresh on the displays if they get stuck (like rasp pis)
- Update the team numbers on the match play screen on other computers when you move to a new match.
- For end-game auto-scored things where it can also be visually checked (like hang levelness), have it show up on the head ref tablet pre-populated so that it can be over-ridden in case the sensor is acting up.
- In elims, list the alliance numbers on the queue display, and on the match play screen, and on the audience display, and on the alliance station displays.
- For 4 team alliances, on the R3&B3 alliance station displays, show the 4th team number in small text in the corner during match intro so that the MC can announce the fourth team during elims. 
- For 4 team alliances, list 4th team on game announcer screen.
- Have the ability to display arbitrary messages on the queuing display (above the upcoming matches). Like "Alliance Selection Beginning Soon - all teams send rep to field" announcements and stuff.
- Automatic method of ensuring luma key is good for all sponsor logos & team avatars. Could be as simple as a batch script to run in the static folder that processes all of the images in the folder. Current script requires knowing the starting color of all the problem colors in the logos in RGB not YCbCr or something.
- Alliance display transition to go from logo w/o background to logo with background without the logo going away

Will probably think of a few others. Thanks!

patfair, 2021-11-01:
Good suggestions! I'll put them on the list for next year.

FletcherS7, 2021-11-01:
> * Some sort of warning issued on the match play screen if a team number is entered that isn't at the event (or isn't on the alliance in elims)

Great idea
But if implemented, the error should NOT be shown during match test.

FletcherS7, 2021-11-01:
Some suggestions I have

- Night mode for Alliance station displays (Display black color), so we can avoid rebooting the displays at night.
- FTA abilities section for the FTA display, this section would include a button to force field reset (and volunteer rest) mode. 
- FTA abilities section for the FTA display, Text that can be displayed on the device screen in order to signal to teams behind the driver station wall. 
Screenshots from FTA Notepad
[image]
[image]
[image]
[image]
[image]
- Add Bandwidth usage per team to Field monitor, FIRST FMS has this and it looks like its reported by the driver station software.
- 2min countdown (Make configurable in the settings) on Head Red display during elims, So the ref can check if a team can use a timeout.
- Standard Ref display (Current Ref display but with only the foul section)
- AV Control panel (Match Play panel but with only the Audience and Alliance display controls)

neyre, 2021-11-02:
Another one:

- Integration with spotify API to tie specific audience display transitions to spotify API actions. For example, if CA could send a "next track" API command whenever switching to "Match Play" and a "pause track" whenever going from final score to no overlay, and "play track" whenever switching to "match intro" that would allow for the DJ role to be mostly automated for quals.

neyre, 2021-11-02:
>     * 2min countdown (Make configurable in the settings) on Head Ref display during elims, So the ref can check if a team can use a timeout.

If implemented, should also display on the match play screen.

FletcherS7, 2022-10-26:
- In elims, list the alliance numbers on the queue display, and on the match play screen, and on the audience display, and on the alliance station displays.
- For 4 team alliances, on the R3&B3 alliance station displays, show the 4th team number in small text in the corner during match intro so that the MC can announce the fourth team during elims.
- For 4 team alliances, list 4th team on game announcer screen.
Added with
https://github.com/Team254/cheesy-arena/commit/9ba5eb6de83f66f1be51a93e6af16bce052dd81d
https://github.com/Team254/cheesy-arena/commit/56ce77f974f00e7f1da7f7f3bf437ec24be16dd7
https://github.com/Team254/cheesy-arena/commit/d9caefca93f5d68a0fca13366825d7f37586b442
https://github.com/Team254/cheesy-arena/commit/362b4a124c1fe436a471700a0385e90b20b659f4
https://github.com/Team254/cheesy-arena/commit/0c199584e4d852f17eb73d9f4686f37092ab8bcc

- Have the ranking display switch to displaying the elims bracket during elims like FIRST FMS.
Added with
https://github.com/Team254/cheesy-arena/commit/edb3aac0807895f0170fc2fc126f2817f3688423

FletcherS7, 2022-11-07:
Add Bandwidth usage per team to Field monitor, FIRST FMS has this and it looks like its reported by the driver station software.
Added with
277e01eebcbe9e8551d27229457381ae3eb1b957