Team254/cheesy-arena #118: Endgame sounds don't play in practice matches

CrispyBacon1999, 2022-04-11:
We're using cheesy arena for an offseason and found that the 30 second warning sound doesn't actually play.

ebot1234, 2022-04-11:
What is your setup? Did the other sounds play as normal?

CrispyBacon1999, 2022-04-11:
We're mostly just using it for scoring and timing, but all other sounds work perfectly fine. Only missing sound is endgame, and even manually pressing it during the match from match test doesn't fix it. (It does work if you press warning outside of a match)

ebot1234, 2022-04-11:
I would check in the static/audio folder to see if the warning.wav is present. If it is, I would try reinstalling CA or building from the source again.

CrispyBacon1999, 2022-04-11:
It does work fine from the field test, and the file is in the right place. Not sure if it's just configured to play at the wrong time or something.

ebot1234, 2022-04-11:
Yeah, that could be it. There are timing settings in the setup tab that can change when that sound is called.

ebot1234, 2022-04-11:
@patfair Any Ideas?

CrispyBacon1999, 2022-04-11:
[image]
Times all look correct to me

ebot1234, 2022-04-11:
Yeah, I'm not sure what would be wrong. I can't reproduce the issues on my end but maybe Pat will have more insight later. 
Did you build CA from scratch or use the precompiled version?

CrispyBacon1999, 2022-04-11:
Precompiled on M1

ebot1234, 2022-04-11:
It might work if you tried rebuilding it using the source. I've fixed problems with my versions of CA by doing so.

CrispyBacon1999, 2022-04-12:
Yeah, I'll give that a try

CrispyBacon1999, 2022-04-12:
Ok, so it seems like it's not an actual issue, that sound effect is just EXTREMELY quiet compared to everything else.

ebot1234, 2022-04-12:
Oh okay, that works too.

closed 2022-04-30 by patfair