Team254/cheesy-arena #130: Event Logo overlaps Audience Display "Bracket" view

FletcherS7, 2022-10-11:
At 100% size the Event Logo will overlap the bracket at most resolutions. 
I tried changing the bracket view CSS to remove the logo all together but was unable to make the transitions from the other views work correctly. 

Would it be possible to replace the Event Logo with the event name at the top of the bracket (Like the dedicated bracket display)
[image]

neyre, 2022-10-11:
Fletcher, what resolution are you running the audience display at? There are some known rendering bugs at non-1080p resolutions that we should probably mention in the CA documention.

FletcherS7, 2022-10-11:
Ratio is 16:9
1200p on this display.
But did have this problem on a 1080p output if the zoom was changed.
The other audience displays have this problem (Besides Alliance Selection getting cut off sometimes)

I guess another fix would be to match the behavior of the event logo on the the Final Score screen.

neyre, 2022-10-11:
I'm going to rename this issue to cover the more general case of incompatibility with other resolutions and zooming. This is definitely a good enhancement that we should pay attention to in the future.

ejordan376, 2022-10-12:
Suggestion, move the Logo to the top right above finals.