Team254/cheesy-arena #163: New substitute teams button commits switch config to fast

FletcherS7, 2023-10-09:
Looks like DCHP config is being sent to the switch to fast and some commands are being dropped.
This only happens when using substitute teams and does not happen when loading a match. Is it possible to the substitute teams button to save the new teams and reload the match (like it had been loaded fresh)?

This also might be something that can be fixed by switching from telnet to SSH for the switch configuration and using a SSH login that defaults to Exec mode.

patfair, 2023-10-09:
I wasn't able to reproduce this problem on the switch that I have (a 3560G). Regardless, I'm adding a 5-second sleep after the command is run and before the switch mutex is released, to prevent simultaneous configs from stepping on each other, in 2d9ca7b864f17ba299ed71d363e64ceb2dafa84b in the Vivid-Hosting AP branch I'm working on.

FletcherS7, 2023-11-13:
Did not encounter this again after the  Vivid-Hosting AP branch was merged.

closed 2023-11-13 by FletcherS7