Team254/cheesy-arena #165: Issues configuring Linksys APs

baconstrip, 2023-10-21:
We tried running an event this weekend with the Linksys APs, but we're having issues configuring the Linksys APs that already have the config loaded.

I've attached the log output of the software configuring the APs, it crashes repeatedly. Just adding a nil check for the field that it crashes trying to read, the software repeatedly fails to configure the AP. 

I'll try to bisect the issue before the end of the weekend, just letting you know. We're currently running 9e03243, which seems to be working.

```
2023/10/20 17:09:54 Listening for driver stations on UDP port 1160
2023/10/20 17:09:54 Listening for driver stations on TCP port 1750
2023/10/20 17:09:54 Connected to PLC at 10.0.100.40:502
2023/10/20 17:09:54 Serving HTTP requests on port 8080
2023/10/20 17:09:54 Access point status:
wlan0     ESSID: "no-team-1"
          Mode: Master  Channel: 40 (5.200 GHz)
          Tx-Power: 23 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -92 dBm
          Bit Rate: unknown
wlan0-1   ESSID: "no-team-2"
          Mode: Master  Channel: 40 (5.200 GHz)
          Tx-Power: 23 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -92 dBm
          Bit Rate: unknown
wlan0-2   ESSID: "no-team-3"
          Mode: Master  Channel: 40 (5.200 GHz)
          Tx-Power: 23 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -92 dBm
          Bit Rate: unknown
wlan0-3   ESSID: "no-team-4"
          Mode: Master  Channel: 40 (5.200 GHz)
          Tx-Power: 23 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -92 dBm
          Bit Rate: unknown
wlan0-4   ESSID: "no-team-5"
          Mode: Master  Channel: 40 (5.200 GHz)
          Tx-Power: 23 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -92 dBm
          Bit Rate: unknown
wlan0-5   ESSID: "no-team-6"
          Mode: Master  Channel: 40 (5.200 GHz)
          Tx-Power: 23 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -92 dBm
          Bit Rate: unknown
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x7a3b20]

goroutine 66 [running]:
github.com/Team254/cheesy-arena/network.(*AccessPoint).configIsCorrectForTeams(...)
	/home/cheesyarena/cheesy-arena/network/access_point.go:198
github.com/Team254/cheesy-arena/network.(*AccessPoint).handleTeamWifiConfiguration(0xc0000df890, {0x0, 0x0, 0x0, 0x0, 0x0, 0x0})
	/home/cheesyarena/cheesy-arena/network/access_point.go:134 +0x160
github.com/Team254/cheesy-arena/network.(*AccessPoint).Run(0xc0000df890)
	/home/cheesyarena/cheesy-arena/network/access_point.go:92 +0x1be
created by github.com/Team254/cheesy-arena/field.(*Arena).Run in goroutine 1
	/home/cheesyarena/cheesy-arena/field/arena.go:670 +0xd6
```

patfair, 2023-10-21:
Oh, if you're using two APs I think I know what the problem is; I only tested with one after my recent set of changes to support the new VH-109 AP, oops.

Try the fix in 7b4d49e6221ee92b536de5da843a6ccbefba7ea9 if you want to and get a chance.

baconstrip, 2023-10-21:
Thanks for the quick response, we'll try after practice today!

baconstrip, 2023-10-22:
Yes, we're using two APs this event. 

Just checked by runngin 7b4d49e , we're still having a similar issue, but it occurs in a different place now:

```
2023/10/22 08:51:06 Access point status:
wlan0     ESSID: "no-team-1"
          Mode: Master  Channel: 36 (5.180 GHz)
          Tx-Power: 23 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -94 dBm
          Bit Rate: unknown
wlan0-1   ESSID: "no-team-2"
          Mode: Master  Channel: 36 (5.180 GHz)
          Tx-Power: 23 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -94 dBm
          Bit Rate: unknown
wlan0-2   ESSID: "no-team-3"
          Mode: Master  Channel: 36 (5.180 GHz)
          Tx-Power: 23 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -94 dBm
          Bit Rate: unknown
wlan0-3   ESSID: "no-team-4"
          Mode: Master  Channel: 36 (5.180 GHz)
          Tx-Power: 23 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -94 dBm
          Bit Rate: unknown
wlan0-4   ESSID: "no-team-5"
          Mode: Master  Channel: 36 (5.180 GHz)
          Tx-Power: 23 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -94 dBm
          Bit Rate: unknown
wlan0-5   ESSID: "no-team-6"
          Mode: Master  Channel: 36 (5.180 GHz)
          Tx-Power: 23 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -94 dBm
          Bit Rate: unknown
wlan1     ESSID: unknown
          Mode: Client  Channel: unknown (unknown)
          Tx-Power: unknown  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x10 pc=0x7a5890]

goroutine 22 [running]:
github.com/Team254/cheesy-arena/network.(*AccessPoint).updateTeamWifiBTU(0xc0000db918)
	/home/cheesyarena/fresh-arena/network/access_point.go:350 +0x2f0
github.com/Team254/cheesy-arena/network.(*AccessPoint).Run(0xc0000db918)
	/home/cheesyarena/fresh-arena/network/access_point.go:95 +0x116
created by github.com/Team254/cheesy-arena/field.(*Arena).Run in goroutine 1
	/home/cheesyarena/fresh-arena/field/arena.go:671 +0x115
```

Thanks again for your help.

patfair, 2023-10-22:
Sorry, I should really refactor the AP code such that it's completely unit-testable. Should be fixed now.

closed 2023-10-22 by patfair

ejordan376, 2023-11-04:
We are still having issues with the APs taking the config properly. It will recofigure them even with a correct config and get stuck in a loop.

ejordan376, 2023-11-04:
We are noticing the network is automatically preconfiguring post-match but when the scorekeeper commits the previous match it reloads the config causing some interment issues with the APs.

patfair, 2023-11-04:
Sounds like an unintended consequence of #94. I'll work on a fix, but if it's really annoying you can try reverting 888b8d468b5a66a709748cc197eb8a36ee2bbc94 in the meantime.

patfair, 2023-11-04:
Fixed in 87b03f27227f3e1a693ed7a841858c6b2e95c79d.

ejordan376, 2023-11-04:
didnt fix, it seems to be crashing when it tries to send the no-team config

ejordan376, 2023-11-04:
still reconfiguring all network, switch and APs just not right at commit match now. we had a green field then lost all DSs, and both Aps get stuck crashing when they the no team commands

FletcherS7, 2023-11-04:
Looks like this issues was not present in 6171f0dfb84f29edbdf368dd3ef71b8c58825064 (This was the commit we ran all of Capital City Classic quals on)

We only switched to 888b8d468b5a66a709748cc197eb8a36ee2bbc94 for play offs at Capital City Classic, So its possible that this change was breaking for quals matches.

commit 6171f0df (2023-10-22): Fix another NPE when using two APs (fixes #165).