Team254/cheesy-arena #192: Unable to Create List for Practice and Ranking Matches

emiran6948, 2024-07-29:
Hello,

First of all, thank you for this FMS you have developed. We are planning an off-season event soon and we will be using your FMS. However, I wanted to create a scenario with random teams, but I am unable to add teams to the FMS interface. Could you please assist me with this issue?

Thank you.

https://github.com/user-attachments/assets/137d7267-5a35-4a05-a984-ec78516fb25d

FletcherS7, 2024-07-29:
Replied on CD
https://www.chiefdelphi.com/t/unable-to-create-list-for-practice-and-ranking-matches-cheesy-arena-manual-scoring/469284/2?u=fletchers7

patfair, 2024-07-29:
I suspect it's an issue with the server not being able to parse the date-time picker output which has been localized to Turkish. Try this fix: https://github.com/Team254/cheesy-arena/commit/f0ff0df7cd8f6690b34b8469d1c5541087135239 in which I'm forcing the locale to `en`.

closed 2024-07-30 by patfair