Team254/cheesy-arena #200: Issuing a 2nd yellow does not automatically promote to a red

thatnerdjack, 2024-10-28:
When issuing a yellow card to a team that already has a yellow card, the card stays as yellow rather than promote them to red. The audience display shows yellow and (per FTA report) also shows yellow on the scorekeeper display.

Suggested behavior:
- Don't allow the head referee to put a yellow card in on the tablet. Only allow to cycle between "no card with yellow outline" and red.
- Allow the head referee tablet to put a yellow card in and automatically promote behind the scenes (this fits more of the in season practice of "score exactly what you see on the field, let FMS take care of the rest").
- Possible scope creep but my preference: make cards just like fouls/tech fouls. Put them in as line items with team assignments, a rule drop down, and a button to cycle red or yellow. This would also allow the head referee to store information as to why cards have been put in.

thatnerdjack, 2024-10-28:
Happy to put in the leg work on this but would appreciate input on what others think is the best approach.

patfair, 2024-10-30:
Right, the intended flow is for the head ref to see that it's yellow and promote to red themselves. That's in line with the Cheesy Arena philosophy of providing maximum control to power users and trusting them to just not do dumb things, but I understand how it would be surprising to someone who is more familiar with FMS.

Agreed that it's probably time for this to change now that more events with FMS-trained volunteers are using Cheesy Arena, and I'll pick it up for next year's version.