Team254/cheesy-arena #206: Failed to connect to Blackmagic device at : dial tcp :9993: connect: connection refused When not configured

cpapplefamily, 2024-11-30:
If not configured should this error not be log?

patfair, 2024-12-01:
It shouldn't do anything if the address field from the settings page is blank:
```go
deviceAddresses := strings.Split(addresses, ",")
for i, address := range deviceAddresses {
	deviceAddresses[i] = strings.TrimSpace(address)
}
```
The above logic probably doesn't handle every non-blank-but-invalid edge case, though.

cpapplefamily, 2024-12-01:
I been watching the CA logs as I been testing and noticed this output at the start and stop of every match. It doesn't seem to check that Blackmagic is being used. Maybe it doesn't cost anything to execute.
[image]

patfair, 2024-12-01:
I fixed the edge cases in the address parsing logic.

closed 2024-12-01 by patfair

cpapplefamily, 2024-12-02:
I'll pull them and watch.

You don't have the Github discussion tab for chats and I have a new
Comment/Question.
Looks like you added a print statement if the Arena.Run loop runs over
3000us.
 log.Printf("Warning: Arena loop iteration took a long time: %dus", time.
Since(loopStartTime).Microseconds())I

I started seeing this error at 3000 to 10000 too and was a bit
concerned until I realized it was microseconds and not milliseconds. In
your opinion what is acceptable and what is a concern? I was running with
NO filed equipment so everything from network configurations are finding
nothing.
Corey

patfair, 2024-12-04:
I would only be concerned if I see that log line repeating consistently; on my machine I only see it maybe once or twice on startup and not again for tens of minutes. When you see it it's probably due to garbage collection or some other process spiking the CPU.

I added it when we were trying to diagnose the driver station protocol error over the summer, so that we can know if Cheesy Arena is ever at fault for not sending packets frequently enough.

commit 2baa45c4 (2024-12-01): Fix gaps in Blackmagic address parsing (fixes #206).