Team254/cheesy-arena #209: Support for other switches

CJBuchel, 2025-02-25:
The Cisco lineup works well enough for the application, but the old 3560G is long out of service. This along with it needing to be purchased second hand makes it a bit of a shot in the dark, at least in my area, I can never find 2 of the same models for redundancy or ease of mind. (Often times these switches come out of datacenters and may have been pushed fairly hard in their lifetime). And buying new cisco switches is not very cost effective or valuable for what you get.

Other similar models work fairly well with some tweaking (i.e, we use a set of 3750G's) but requires a fair amount of reconfiguring. And on rare occasions, modifications to the Go code. In our case, it was just a tweak to the config script. But due to Cisco shinanigins, it also runs on older firmware, with no easy way of updating the firmware to something later, i.e v15.0 used in the cheesy configuration.

Recently I've thought more and more about trying to get different switches to work, like TP-Link and Ubiquity which can be bought new for a reasonable price with modern hardware and software, fairly easily upgradible and in the case of Unify, also managable through a controller. In our case Unify products seem to work well for our needs, as we have a few Unify AP's for team internet and admin/FMS networking. And a dream machine for the gateway, also acting as the unify controller for site management. Which makes quick venue setups fairly easy to configure all the networking at the start of the day, without needing to individually touch the switches or ap's scattered around the venue. Especially for changes throughout the event.

In the coming months I was going to play around and see if I could integrate the Unify API into cheesy
https://ubntwiki.com/products/software/unifi-controller/api
Unify switches also have the same SSH style as cisco, but it seems more flexible to use the top layer API from the controller, and with the Radio now using an API, having the switch also using an API just seems nice to abstract away the nitty gritty.

It got me thinking that it would be nice to see officially in Cheesy as well, perhaps a dropdown for the type of switch/networking in the setup page to allow for different support.

Curious on the thoughts about it, I'll be trying it out in my own spare time for Unify. But I thought I'd chuck it down as a Feature request/Issue.

CJBuchel, 2025-05-28:
Ah, funny you should mention it.
I did recently get the Ubiquiti system working within my own tests, it hasn't been fully cleaned up yet but the configuration works quite well. I haven't tried to do any multi selector vendor in case the ubiquiti system broke and I needed a backup (currently hard coded for now).

But the Ubiquiti method works surprisingly well, it's all API based and I've noticed a significant speed different when configuring the switches. But it does have both the good and bad requirements of a unifi console/unifi virtual controller. In our case it was fine because we have an offseason box, which includes a few ubiquiti switches and a UDM Pro as the gateway router for internet connections.

It does allow the configuration to be fairly flexible, as the FMS only ever needs to communicate to the main unifi controller never any switches directly, whether it be running in a virtual container, an rpi or a proper UDM. And all changes propogate to the rest of the unifi network. In theory you could setup multiple unifi switches all over the field with tagged networking ports. Such as 10,20,30 etc... And then all changes from the FMS would propagate to them just as easily. It offloads the specific configuration from the FMS code to instead the unifi controller, making it easier to completely rebuild the network depending on the event. And all the switches (if tagged correctly) would still function as expected. For instance a single score switch taking in all team driverstations, or dual switches for red and blue alliances.

The only issue is that it is more complicated to setup initially, especially because it needs some kind of unifi controller. Virtual or physical. Which may drive up costs for some. I'll share a snippet of the code I changed up for the unifi system. It's not fancy but it gets the job done in our scrimmage tests. And instead of calling the standard cisco config, I instead call this one.

```go
package network

import (
	"bytes"
	"crypto/tls"
	"encoding/json"
	"fmt"
	"io"
	"net/http"
	"net/http/cookiejar"
	"strings"
	"sync"
	"time"

	"github.com/Team254/cheesy-arena/model"
)

// const (
// 	switchConfigBackoffDurationSec = 1
// 	switchConfigPauseDurationSec = 1
// 	switchTeamGatewayAddress       = 4
// )

type UnifiNetwork struct {
	address string
	password string
	csrfToken string
	mutex sync.Mutex
	configBackoffDuration time.Duration
	configPauseDuration time.Duration
	Status string
}

var UdmIpAddress = "10.0.100.1" // The UDM is the gateway router
var authRoute = "api/auth"
var defaultRoute = "proxy/network/api/s/default"
var restRoute = fmt.Sprintf("%s/rest", defaultRoute)
var statRoute = fmt.Sprintf("%s/stat", defaultRoute)

func NewUnifiNetwork(address, password string) *UnifiNetwork {
	return &UnifiNetwork{
		address:                 address,
		password:                password,
		configBackoffDuration:   1 * time.Second,
		configPauseDuration:     1 * time.Second,
		Status:                  "UNKNOWN",
	}
}

// Login for the UDM Router
func (ubnt *UnifiNetwork) login(client *http.Client) error {
	// Create login request
	loginData := map[string]string{
		"username": "root", // Replace with your username
		"password": ubnt.password, // Replace with your password
	}
	jsonData, _ := json.Marshal(loginData)
	
	req, _ := http.NewRequest("POST", 
		fmt.Sprintf("https://%s/%s/login", ubnt.address, authRoute), 
[... 285 lines cut ...]
			ubnt.Status = "ACTIVE"
			return nil
		} else {
			ubnt.Status = "CONFIGURING"
		}
		time.Sleep(time.Second)
	}

	// If we reach here, the device is not online
	fmt.Println("Network config timeout")
	ubnt.Status = "ERROR"
	return fmt.Errorf("network config timeout")
}
```

But something else I thought about recently is how the official FMS configures networks. 
https://fms-manual.readthedocs.io/en/latest/fms-whitepaper/fms-whitepaper.html

Unlike in cheesy where the scoreswitch both has tagged ports and is reconfigured for those ports to provide a DHCP server that hands out IP addresses for each VLAN/team. The actual FMS as far as I can tell only has the ports tagged, similar to the SCC switches. And I believe the FMS server itself hosts the DHCP servers for all 6 teams. 

To me this sounds like a very clean and flexible solution. As it would likely be compatible with any L3 switch that could tag the ports without any cheesy code change. And depending on how it was implemented, could offload the config as a proper code configuration by the FMS or API. Likely making it quite a bit quicker than the normal Cisco switch updates. And probably would provide more information back for errors if any.

It's a neat idea but I've not looked at it much. I'm also unsure on the best approach to do it, I've briefly looked into it and in some cases it seems like it could be built into the cheesy code using a DHCP server package or a custom written one. But I've also seen In other cases unrelated to the FMS people have run a small rpi or linux computer hosting a DHCP server, and on the same machine serving a small API server to access and configure that DHCP server to work. 

Although our setup is currently done in a proxmox server where the FMS is an LXC container. And I'm unsure how well this method would work virtualised as another container possibly that the FMS could talk to using an API. Still, it's a cool idea and one that I'll definetly try to look into eventually. But for now unifi is our current setup for our August event, but maybe afterwards I can have a crack and share my findings.

CJBuchel, 2025-05-28:
True, I don't expect it to be an easy setup nor very feasible built into cheesy. And if anything would probably be a separate program running on a pi or pc which cheesy would access via an API. But even then it's an extra component which would be needed for an FMS setup.

My thinking is that while it doesn't eliminate the layer 3 switch requirements, it does mitigate code changes needed when you use a different switch. Most events will need to configure a switch prior to the event anyway, including setting up the tagged ports, the ACL's and any other special needs they might have. But the pre-match configuration sent by the FMS would no longer be locked to a Cisco switch. So an event could have options as to what switch they wanted to use, and it would likely work without any code changes to Cheesy. As even some Cisco switches other than the 3560G don't fully work without code changes to the underlying commands sent from Cheesy.

It's a bit of niche issue, but it's food for thought. As I think adding an extra bit of code for every brand of switch in a dropdown while would greatly be beneficial. Likely would also get complex depending on the amount of switches one would try to support.

In an ideal world for me, I'd have a dropdown for 3 options. `Cisco (default)`, `Unify` and then `DHCP API`.

FletcherS7, 2025-05-28:
Any Cisco switch running IOS 12 or newer (with a IP-Base or higher license) will work with the commands sent from Cheesy Arena, that should cover _any_ Cisco layer 3 switch made after 2005.
If you are trying to use something made before 2005 I suggest you reconsider.

FletcherS7, 2025-05-28:
Correction.
Any switch made after 1999.
I misremember when IOS 12 released.

I would be curious what code changes you had to make in Cheesy Arena.
Changes to the IOS startup config is expected since interface numbering is very switch specific., but you should not need to modify the commands in Cheesy Arena.

CJBuchel, 2025-05-28:
Hm, hang on. I'll try see if I have the old configs around. 
It was a few years ago with a standard 3750G model.
And then an 8 port 3750G variant which we were trying to use for a small portable scrimmage box in a briefcase.