Team254/cheesy-arena #211: Add support for Bitfocus Companion

FletcherS7, 2025-03-11:
Add support for Bitfocus Companion (https://bitfocus.io/companion)
Enable sending "events" to the Companion software such as, match preview, match play, post result, timeout, etc.

Companion can then be used to trigger any of the vast array of devices it supports. Enabling far reaching automation support without building direct integrations into Cheesy Arena.

(Feature Parity with FIRST FMS)

ithompson, 2025-05-24:
Notes on how to implement this:

* Bitfocus remote API is on TCP port 51234 or UDP port 51235
* To signal an event, dial the server, send the following string, and close the socket:
  * `LOCATION <page>/<row>/<column> PRESS\n`
  * eg `LOCATION 1/3/5 PRESS\n` to trigger page 1 row 3 column 5
* Remote API only supports signaling events by position on the screen, cannot use abstract names for events
* Would need a configuration page added with the following:
  * Bitfocus companion server address/port
  * Page/Row/Column bindings for these events:
    * Set match overlay
    * Match start
    * Teleop start
    * Endgame warning
    * Match end
    * Results posted
    * Alliance selection

FletcherS7, 2025-05-24:
Here's what the configuration looks like in the FIRST FMS Audience Display software

[image]

And here's the relevant page from the FMS Manual
https://fms-manual.readthedocs.io/en/latest/audience-display/automation/bitfocus-companion.html 

@ithompson Also would want an event for Match Abort.